dwyl/learn-raspberry-pi #10: StereoPi

nelsonic, 2019-01-28:
http://stereopi.com
https://www.crowdsupply.com/virt2real/stereopi
Via: 
https://www.raspberrypi.org/blog/compute-module-3-on-sale-now-from-25/
https://news.ycombinator.com/item?id=19016100

realizator, 2019-02-01:
@nelsonic Hi! Occasionally found this issue with a mention of our project StereoPi. If you have some questions - I'm ready to answer! :-)

nelsonic, 2019-02-01:
@realizator I will have _lots_ of questions very soon. 🔜 
My _first_ question is: what are the timelines for shipping StereoPi boards?
(_e.g: if I order a couple of them today, when can I expect them to arrive?_)

realizator, 2019-02-01:
@nelsonic This info is in a pledges description (above yellow button). We expect to start delivery at the end of March. If some miracle happens and we will collect funding goal earlier (not in a 40 days), then CrowdSupply can transfer money to our account earlier and we will start production earlier too. SO it partialy depends on bakers activity :-)

nelsonic, 2019-02-01:
@realizator thanks for confirming. I will place my order this afternoon when I get [@home](https://github.com/dwyl/home). 
(_I don't make internet purchases on my "work" computer ..._)

realizator, 2019-02-01:
@nelsonic I read a bit about dwyi and very impressed! One of StereoPi project ideas was to share with other people a great tool for making and learning something new, with a comfort, ease and affordable price.
May be it will be interesting for you to cooperate with our project in a several ways? Here are some ideas:

1. I still have several prototypes from our first 20 pcs batch, and I can send you one. If you will be able to test it, write some examples or manuals, and some reviews than it is great! It will help for other peoples to start with StereoPi when they receive their boards, and also may help to our crowdfunding campaign (which is the only way to bring project to live now).

2. In the nearest time we plan to start filling up our Wiki, and we need help with it. We will need several things here:
- fixing our English, as we're not native speakers
- help with review and polish articles
- add new how-tos and step-by-step guides

P.S. I see your next comment while writing my answer. I will be pleased for your order :-)

realizator, 2019-02-01:
@nelsonic You can write me email: [my Github nickname] [at] gmail [dot] com

nelsonic, 2019-02-01:
[image]

This is what I'm thinking of getting as our "starter pack" for @home so we can start experimenting: 
[image]

realizator, 2019-02-01:
@nelsonic Thank you for your support!!!

nelsonic, 2019-02-02:
@iteles given that the shipping date for the StereoPi is only in April 2019
[image]

Do you think it's prudent to get it shipped to @home address or to Carca?

realizator, 2019-02-02:
> For the full pledge, let's figure it out today because we're going to get hit with some insane import duties here if they ship from the US 😕

I don't know your customs rules. But if it helps you CrowdSupply can send 4 different parcels, not one big. Or to 4 different addresses. If you want some other ideas - let me know, I can discuss it with CrowdSupply team.

iteles, 2019-02-02:
@realizator Thanks for the flexibility, that may well help! I'll take a look at the import duties and we'll get in touch 😊

nelsonic, 2019-02-07:
in case you are wondering why our order has not yet been placed ... credit card payment declined: 😕 
[image]

This is **_super_ annoying** considering we _just_ used the card to make other online purchases _today_!
And we need to use it to pay for a few more items elsewhere ...
This is being investigated. ⏳

realizator, 2019-02-07:
@nelsonic If you faced this issue with your card on CrowdSupply only, and it repeats several times, I can contact CS team to help to resolve this situation. Just tell me, I'm here!

nelsonic, 2019-02-07:
@realizator I tried several times over the last few days.
Each time it fails the checkout form wipes my data and I have to re-enter everything. (_super frustrating_)

I have just used my _personal_ credit card to complete the transaction:
[image]
Ordered two kits (_one "Starter" and one "Deluxe"_):
[image]
(_removed other accessories from order as my personal credit card has a low online spending limit ..._) 🙄 

I don't know if Crowd Supply give you _access_ to our shipping address,
let us know if they don't.
Thanks!

realizator, 2019-02-07:
I gave a link to this issue, they will look at this question.

nelsonic, 2019-02-07:
Email Sent: 📥 
[image]

_Really_ Looking forward to trying out the StereoPI! 📷 🤖 
Thanks! ✨

nelsonic, 2019-02-11:
@realizator _congratulations_ on reaching your crowd funding goal _early_! 🎉 
[image]

realizator, 2019-02-11:
Thank you Nelson!!!!
I'm glad we will share the StereoPi with this world! :-)
Occasionally this day of 100% funding coincides with my wedding
anniversary... I think it is a good sign! :-)

Regards,
Eugene

realizator, 2019-02-28:
@nelsonic My tracking service says parcel already arrived to your PO from 22 of February, so you can pick it up.